jamescarruthers/PlotteRXY #1: STEP Files Available?

ScottEgan, 2023-02-16:
Hi,

Awesome project, thanks for all the detail! 

I'm in the process of modifying my plotter to be corexy and would like to use some of your parts with slight modifications. Do you have the STEP files available?

Thank you!

jamescarruthers, 2023-02-23:
Hi,

The project is designed in Fusion 360 — so I don't think STEP is available.

James

ScottEgan, 2023-02-23:
Hi James,

Thank you for the reply. Is it possible to export the native Fusion files then? I should be able to handle converting them.

TacyPW, 2024-03-15:
it would be really nice to have the fusion files in the repo as well

lambda1103, 2024-06-17:
Just to second (or third :D) this, I'd also appreciate it if you could share the .f3d file. Would be very helpful for assembly and small modifications.

theredwonder, 2024-09-03:
Wanted to also add an upvote for native CAD files. A ZIP of the Fusion360 assembly would be huge! All of the parts in STEP format would also be super helpful. 
https://www.reddit.com/r/Fusion360/comments/1691l3g/export_formats_fusion_360_personalfree/

Looking to start a build on this soon, but am starting with a slightly different aluminum extrusion. Would love to tweak to fit. Thanks!

mattkerrison, 2025-03-03:
@jamescarruthers any chance of an update here? Either the f3d file or step files? I have a heap of Vslot left over from a cnc build but a lot of it isn’t 2020 or 2040 so it’d be good to adapt the 3d printed parts