jamescarruthers/PlotteRXY #2: picture of whole plotter

TacyPW, 2024-03-12:
would it be possible to post a picture of the entire plotter as an additional reference for the overall layout

jamescarruthers, 2024-03-12:
> would it be possible to post a picture of the entire plotter as an additional reference for the overall layout

Sure! I really need to get around to it. I was put off not having a pretty environment to take them in - but I have that now.

TacyPW, 2024-03-15:
yay! i didnt know if you'd respond - but yeah it would be super helpful as a complement to the diagram that you made

mationai, 2024-11-09:
Second on picture (better if video) of the build. Really want to see it.

kaosbeat, 2025-05-11:
I'm building one, I'll post some pics here...
Currently building in 3D, I can share the .blend file too

[image]

[image]