jamescarruthers/PlotteRXY #4: FluidNC Config File not compatible with latest version of fluidnc

TacyPW, 2024-04-26:
The way that fluidnc wants uart channels to be set up has changed, see [this page](http://wiki.fluidnc.com/en/config/uart_sections) of the wiki. 

> Trinamic UART devices like TMC2208 and TMC2209 no longer support the subordinate "uart:" subsection (making that work was just too hard); you must use the external UART section with "uart_num: N":

the solution is to put the uart definition at the top level of the config file and then refer to it in each motor config. 

this is uart section:
```yaml
uart2:
  txd_pin: gpio.22
  rxd_pin: gpio.21
  baud: 115200
  mode: 8N1
```
and this is one of the motor configs:
```yaml
 motor0:
      limit_pos_pin: gpio.34:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        uart_num: 2
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26:high
        disable_pin: NO_PIN
```
i have this up and running on my machine and it seems to work just fine.

TacyPW, 2024-04-26:
for reference here is my whole config file: 
```yaml
name: "PlotterXY"
board: "FYSETC E4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 2
  pulse_us: 2
  disable_delay_us: 0
  segments: 20

kinematics:
  corexy:

uart2:
  txd_pin: gpio.22
  rxd_pin: gpio.21
  baud: 115200
  mode: 8N1

# 20t pulley
# steps per mm/microsteps   80/16   40/8    20/4    10/2    5/1
# resolution in mm          0.0125  0.025   0.050   0.100   0.200

# 16t pulley
# steps per mm/microsteps   100/16  50/8    25/4    12.5/2  6.25/1
# resolution in mm          0.010   0.020   0.050   0.080   0.160

# 1/4 microsteps gives best torque for a wide range of speeds

# Board stepper driver address and gpio
# axis  add   step  dir
# x     1     27    26
# y     3     33    32
# z     0     14    12
# e     2     16    17

axes:

  x:
    steps_per_mm: 20
    max_rate_mm_per_min: 24000
    acceleration_mm_per_sec2: 2400
    max_travel_mm: 431
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 431
      positive_direction: true
      seek_mm_per_min: 4000
      feed_mm_per_min: 600
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.34:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        uart_num: 2
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26:high
        disable_pin: NO_PIN

  y:
    steps_per_mm: 20
    max_rate_mm_per_min: 24000
    acceleration_mm_per_sec2: 2400
    max_travel_mm: 279
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 279
      positive_direction: true
      seek_mm_per_min: 4000
      feed_mm_per_min: 600
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        uart_num: 2
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.14
        direction_pin: gpio.12:high
        disable_pin: NO_PIN

  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 64000
    acceleration_mm_per_sec2: 4800
    max_travel_mm: 5
    
    motor0:
      solenoid:
        output_pin: gpio.2
        pwm_hz: 30000
        off_percent: 0.000
        pull_percent: 100.000
        hold_percent: 40.000
        pull_ms: 50
        direction_invert: false

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

# run a fan off the 24V/5V fan header – switch on and off with the WebUI mist button or by issuing an M8 command, turn off with M9
coolant:
  mist_pin: gpio.13
  delay: 0

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: true

```

jamescarruthers, 2024-05-05:
Thank you, Tacy

Wauter, 2025-06-06:
I wasn't able to get this going with the yaml file in the repository, the x axis still has the uart section. 

now, I just managed to get is going using the x and z stepper motor sockets - its quite likely maybe my board is defective on the y axis. but the config also seems a little smelly for y - still i could not get it going on my board after trying to correct.

this information suggests the config file is not correct for Y, but I still wasnt able to get it going with these details.
http://wiki.fluidnc.com/en/hardware/3rd-party/fysetc_e4 -> the fluidnc documentation suggests the y uart address is 2, not 0
https://github.com/FYSETC/FYSETC-E4 -> the E4 documentation says that Y-Step	is gpio 33 and Y-DIR is 32

I also identified a couple of changes to the config format (fluidnc version 3.9.7):
mist pin has a different syntax now. can add home current to config

This is my working config using the x and z steppers: (working as in the steppers turn while homing during bench wire test, i turned down the amperage on the motors for now to avoid overheat)

I would make a pull request but probably better to figure out the y issue first, if someone else could try/confirm with their setup..

`name: "PlotterXY"
board: "FYSETC E4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 2
  pulse_us: 2
  disable_delay_us: 0
  segments: 20

kinematics:
  corexy:

# 20t pulley
# steps per mm/microsteps   80/16   40/8    20/4    10/2    5/1
# resolution in mm          0.0125  0.025   0.050   0.100   0.200

# 16t pulley
# steps per mm/microsteps   100/16  50/8    25/4    12.5/2  6.25/1
# resolution in mm          0.010   0.020   0.050   0.080   0.160

# 1/4 microsteps gives best torque for a wide range of speeds

# Board stepper driver address and gpio
# axis  add   step  dir
# x     1     27    26
# y     3     33    32
# z     0     14    12
# e     2     16    17

axes:

  x:
    steps_per_mm: 20
    max_rate_mm_per_min: 24000
    acceleration_mm_per_sec2: 2400
    max_travel_mm: 860
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 860
      positive_direction: true
      seek_mm_per_min: 4000
      feed_mm_per_min: 600
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.34:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          baud: 115200
          mode: 8N1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26:high
        disable_pin: NO_PIN

  y:
    steps_per_mm: 20
    max_rate_mm_per_min: 24000
    acceleration_mm_per_sec2: 2400
    max_travel_mm: 600
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 600
      positive_direction: true
      seek_mm_per_min: 4000
      feed_mm_per_min: 600
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2209:
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.800
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.14
        direction_pin: gpio.12:high
        disable_pin: NO_PIN

  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 64000
    acceleration_mm_per_sec2: 4800
    max_travel_mm: 5
    
    motor0:
      solenoid:
        output_pin: gpio.2
        pwm_hz: 30000
        off_percent: 0.000
        pull_percent: 100.000
        hold_percent: 40.000
        pull_ms: 50
        direction_invert: false

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

# run a fan off the 24V/5V fan header – switch on and off with the WebUI mist button or by issuing an M8 command, turn off with M9
coolant:
  mist_pin: gpio.13
  delay: 0

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: true
`