jamescarruthers/PlotteRXY #6: What length of bolts and how many?

bushmango, 2024-10-08:
From this segment of the bill of materials, I'm assuming these are the D5M4 bolts that you linked to on Aliexpress.

The quantity and length of the bolt aren't listed and I'm not sure what they are for. 
Do you need one for each wheel?

I ordered the rest of the parts except for this and the 2.5mm bolt for linear guide which I also don't know the length of.

<<
5-M4 Shoulder bolts |   |   |   | 5mm shoulder, 4mm thread - I recommend these over normal bolts as I found some bolts are not straight - causing issues with wheels being misaligned | https://www.aliexpress.com/item/1005003510534996.html
>>

I'll happily put in a pull request with an updated bill of materials when I get mine assembled.

Thanks in advance! I'm excited for this project

jamescarruthers, 2025-05-08:
Sorry for the delay in reply - they're for the wheels.

I hope you got it sorted in the end.

kaosbeat, 2025-05-08:
I think they are supposed to go for example here 

[image]

If I unbderstand the allieexpress link from the BOM correctly

[image]

1mm+4mm+6mm+12mm+6mm+4mm+1mm = 34mm

The only thing that really matters is L meaning for the wheels in the screenshot I would order the 35mm lengths and put some extra washers if needed.

Is this correct?

found some more minor errors in the BOM (this one https://github.com/jamescarruthers/PlotteRXY/issues/3 )
I can continue my work and keep track of the mistakes.. I'll report back.

jamescarruthers, 2025-05-08:
Yep! That's exactly it.

Just found them in my emails - I ordered 10 x 35mm and 5 x 30mm (it was 5 as that was the min quantity)