jamescarruthers/PlotteRXY #8: solenoid voltage

kaosbeat, 2025-05-11:
I'm building the plotter with the solenoid option.
The solenoid from the BOM is 12V, I'm working with the [FYSETC-E4](https://github.com/FYSETC/FYSETC-E4) as suggested and 24V voltage as the main power source.
Where do I connect the solenoid? and is it safe to use the 12V solenoid on 24V?

Wauter, 2025-06-06:
Hey, not sure you still need it. 

if you use the 12V solenoid with 24V, the current doubles from its intended rating. This can lead to overheating / melting in extreme case. 

You are best getting a 24V rated solenoid, but for initial testing you can also:
 - just power the E4 with 12V, it will take it fine. Maybe your steppers will be less powerful, not too sure. (I am powering mine with 12 V to start with)
 - change the solenoid config setting to lower the duty cycle:
        off_percent: 0.000
                pull_percent: 100.000 -> change this to 100.000
                hold_percent: 40.000 -> change this to 20.000
        the effect of this is that by PWM, you will reduce the current by half, bringing it in line with the 12V spec. 
however, if the controller for whatever reason gets stuck on 100% duty cycle you might have fire risk.. would not trust it except for testing.

kaosbeat, 2025-06-07:
Working with a lower duty cycle, for now no problem during prolonged use .
It does get burning hot, need to take care not to touch when swapping pens. For now doing supervised use only.
It works really well! Very happy with the build, but I'll look into swapping the solenoid

jamescarruthers, 2025-06-07:
In reality it doesn't really matter running at 24V.

You shouldn't need to run it at full power if it's setup right - the very short pull current is the biggest one, and then the hold should be a lot smaller.

Essentially - it runs cool.