jamescarruthers/PlotteRXY #9: V wheels size required?

Wauter, 2025-05-12:
Hi, 

I am currently looking into collecting the parts to build one of these. 

one question: 

the required size for the V wheels, the ali link is dead for me. I gather, from the bolts, that the ID is 5mm (4.9?), but I am not sure the size of the wheel, when browsing around,  I see 'small' 15.5 and 'big' 24.8 on offer ( https://www.aliexpress.com/item/1005004275676552.html )

do the wheels have to be a specific material? I assume hard plastic, like pom would be good?

---

some observations from a new set of eyes looking at the materials, just in case it is useful areas where things were maybe a bit unclear on first read.  (I think that is super fine BTW, and I appreciate  sharing the project in any form - just sharing in case this is useful to the project)

link for limit switches is dead - i assume plain old standard limit switches, which I have at hand

wasn't sure about the description of the v nuts, but I think I did figure it out. -> these plain old ones? https://www.aliexpress.com/item/32853205697.html

I struggled with the shoulder bolts, somehow - i am blind, since the required info is there now (I had missed the length), but I found the info in the related ticket on wiki. _(only now I opened the nuts and bolts section and behold it is all neatly there actually, whoops might be nice the have a little message in the first sheet about that for dummies like me..)_

would be great to have more angles on a finished printer for reference, but should be fine how it is as well.

kaosbeat, 2025-05-15:
Hey,

I put everything in 3D to check the sizes. About the bolts: https://github.com/jamescarruthers/PlotteRXY/issues/6#issuecomment-2863556965

I'm starting my built today, I'm expecting the parts later today so I will try and keep track of my progress

The wheels I bought are 24mm diameter, hopefully that is the right size

The limit switches, I'm still unsure what is the best approach, I have some from an old 3D printer. will test

Wauter, 2025-05-20:
great, good luck! I also ordered parts, only held off on the wheels for now and the solenoid. 
first batch of parts arrived today, going to start printing this week.

jamescarruthers, 2025-05-20:
Good luck both of you!

The limit switches are just the typical standard size ones like this:

https://ooznest.co.uk/product/micro-limit-switch/

Wauter, 2025-05-21:
for anyone that needs to read this: regular 2020 profile is not the same as V-slot 2020 profile.. whoops

closed 2025-06-06 by Wauter