jamescarruthers/PlotteRXY #11: 4WD parts mounting schematic / belt

kaosbeat, 2025-05-23:
I'm struggling with how the belt fits the 4WD option. The normal side is exactly the same as the 2motor option, but how does the belt move at the opposite side?
I think I need to cross the belts like this?
[image]

and then?

[image]

and

[image]

This gives me a problem. The motor doesn't fit well and the bearings don't have enough free space because of the motor

[image]

Any help appreciated

kaosbeat, 2025-05-23:
apparently I printed one of the 4WD parts mirrored.. still dazzled though

is this how it should be?

[image]

jamescarruthers, 2025-05-23:
I'll take a photo - I need to add it to the repository anyway.

kaosbeat, 2025-05-26:
not yet operational, but taking shape

[image]

the 4WD is still unclear

[image]

I did it like this, but I cannot mount the corner screws because of the pully on the motor side.

I can off course modify the 3D and reprint, but it feels like I made a mistake ?

jamescarruthers, 2025-05-26:
Yep, you need to use flat head screws.